felHR85/UsbSerial #160: Usb and car radio

jefferson, 2018-08-04:
Is possible to connect android on car for playing music through UsbSerial?

felHR85, 2018-08-06:
Hi @jefferson 
If that radio has a Serial interface or a CDC Usb interface it should be possible

jefferson, 2018-08-07:
Hi @felHR85 
Thanks for your answer.

My car has a Usb Interface but I don't know if is "CDC Usb".
I tried something for get wich protocol my car use but dont's works to.

Is it possible to test to see what protocol my car uses?

felHR85, 2018-08-08:
Steps I would follow:
- Check information about that particular radio to know more info.
- Connect the radio to your PC and using PuTTY monitor what is happening.
Maybe it uses the CAN protocol (used a lot in automotive industry) but it's hard to say

jefferson, 2018-08-08:
Thank, I'll test the steps. 
I think the car should use [Mass Storage class (MSC device) ](http://www.kenwood.com/jp/usb/english.html) to fetch the files and play.

closed 2018-10-28 by felHR85