felHR85/UsbSerial #161: Failed to resolve: com.github.felHR85:UsbSerial:4.5.2

makarna35, 2018-08-05:
**From  README.md:**
Gradle

Add the jitpack repo to your your project's build.gradle at the end of repositories

/build.gradle

allprojects {
	repositories {
		jcenter()
		maven { url "https://jitpack.io" }
	}
}

Then add the dependency to your module's build.gradle:

/app/build.gradle

compile 'com.github.felHR85:UsbSerial:4.5.2'

--------------------------------------
**But I ger this error in Android Studio:
Failed to resolve: com.github.felHR85:UsbSerial:4.5.2**

makarna35, 2018-08-05:
Sorry.
This error Cause of me. I need insert maven { url "https://jitpack.io" } to allprojects. I'm insert before buildscript.

closed 2018-08-05 by makarna35

Greenscreener, 2018-08-06:
Had the same problem.