felHR85/UsbSerial #167: ESP8266, OTG cable AT+CIPMODE? command can't get answer.

makarna35, 2018-10-02:
Hi;
I'm test this with one more android device and one more ESP8266 module. I use ESP8266, HL-340 converter and OTG cable.
If I'm send AT+CIPMODE? command to ESP8266 I can't get answer. it seems like a UsbSerial BUG

If I'm send any other command to ESP8266 after send AT+CIPMODE? command (Ex: AT+CWMODE?) I get two answer with include AT+CIPMODE? answer.

I'm test it with example sync (https://github.com/felHR85/UsbSerial/tree/master/examplesync) result is same...

makarna35, 2018-12-17:
I test it with DroidTerm app but I can't use DroidTerm app I can't accsess to ESP8266

nkhs, 2019-03-25:
I am working with ESP8266 for fimrware Update.
I converted https://github.com/espressif/esptool/blob/master/esptool.py to Android(java)
For serial communication, I am using https://github.com/felHR85/UsbSerial.

I am using https://github.com/felHR85/UsbSerial/tree/master/examplesync.
With FTDI, there is no problem.
But, with CP2102, there is problem.
The esp8266 cannot be firmware bootloader mode(https://github.com/espressif/esptool/wiki/ESP8266-Boot-Mode-Selection#select-bootloader-mode).