felHR85/UsbSerial #183: Problem with definition of variable in examplemultipleports

W1il, 2018-12-20:
Hi, I want to modify the code of the examplemultipleports so that the parameters of the serial port can be variables that I can modify later, I have tried to declare it as variable (mConfigDataBit = DATA_BITS_7) but it always throws me error, someone can tell me what type of variable is DATA_BITS_8, DATA_BITS_7, DATA_BITS_6, DATA_BITS_5?

Here is my code:
`private int mConfigBaudRate;
    private int mConfigDataBit = DATA_BITS_7 ;
    

    private final BroadcastReceiver usbReceiver = new BroadcastReceiver() {
        @Override
        public void onReceive(Context arg0, Intent arg1) {
            if (arg1.getAction().equals(ACTION_USB_ATTACHED)) {
                boolean ret = builder.openSerialPorts(context, mConfigBaudRate,
                        UsbSerialInterface.mConfigDataBit,
                        UsbSerialInterface.STOP_BITS_1,
                        UsbSerialInterface.PARITY_NONE,
                        UsbSerialInterface.FLOW_CONTROL_OFF);
                if(!ret)
                    Toast.makeText(context, "Couldnt open the device", Toast.LENGTH_SHORT).show();
`
Regards

felHR85, 2018-12-20:
@W1il What kind of error is throwing?

W1il, 2018-12-20:
Hi felHR85, 

Thanks I already solutions that problem, now I have another, when I open the application and connect the USB cable, the application works but if for some reason I disconnect the USB cable and again connect it the application no longer runs for more attempts to make connect and disconnect the usb cable until the application finishes and again open it, there is some way to change this so that each time the USB cable is connected the application is reinitialized (both the MainActivity and the UsbService)