felHR85/UsbSerial #190: Device Not Supported (Intermittent)

anthonywob, 2019-01-07:
I'm not sure if this is the correct place to ask this, if not please close and point me in the right direction.

I have built my app with the example code as the method to initiate the serial connection and it works most of the time.

The issue I am having is that sometimes on a reboot or if the USB Serial Adapter is unplugged and then plugged in again I get the "USB device not supported" message. It can be corrected by unplugging and plugging it back in either once or a couple times. This happens intermittently with 3 different USB to Serial Adapters Chipi-X10 (FTDI Chipset), PROLIFIC PL2303HX and a generic cable I believe has a FTDI chipset.

My Question: Is it possible to programmatically try to reconnect the device without unplugging the physical USB device? I can't seem to find a method to do this.

Thanks in advance and great work on this library.

mws-rmain, 2019-01-11:
Check #193.
One of my test devices (a Panasonic ToughPad FZ-B2) had a embedded USB-connected WiFi adapter. When android supplied a list of connected USB devices, they appear in random order. With the possible exception of the multi-device code, the examples in this repo only attempt to link to the first (non-hub) device in the list. If this was the USB-WiFi in my tablet, I'd get 'not supported'. If it happened to be the serial adapter, everything worked fine.

Hopefully this is also your issue, and #193 will fix it for you too.

anthonywob, 2019-01-15:
@mws-rmain Thanks for the response I'll check that out and report back.

anthonywob, 2019-01-15:
@mws-rmain I think you nailed it. Been trying to reproduce the error but cant seem to at this time. Connects everytime. Thanks! 
Awesome library and awesome work on this fix I saw it was committed a few days ago.

I'll close this issue, thanks again!

closed 2019-01-15 by anthonywob