felHR85/UsbSerial #200: App stopped when creatUsbSerialDevice

Aromboy, 2019-01-22:
Hi, first of all , I´m a little newbie with this.
My problem is that when I create a new object of the UsbSerialDevice class in a Fragment and run the app, then when I open that fragment with a navigation drawer it crash and says it stopped.
I did´n write pretty much: 
 
    

    public class EstadoFragment extends Fragment {

       private UsbDevice device;
       private UsbDeviceConnection usbConnection;
       UsbSerialDevice serialDevice = UsbSerialDevice.createUsbSerialDevice(device,usbConnection); //this line makes it crash and i don´t know why

       private boolean supported = serialDevice.isSupported(device);

       @Nullable
       @Override
       public View onCreateView(@NonNull LayoutInflater inflater, @Nullable ViewGroup container, @Nullable Bundle savedInstanceState) {
           View view = inflater.inflate(R.layout.fragment_estado,container,false);

           if (supported != false){

               Toast.makeText(getContext(),"supported",Toast.LENGTH_LONG).show();

           }

           return view;

       }
    }

I also tried putting everything in the onCreateView but it does the same. Probably this is an easy thing to resolve but i can´t figure it out how.I have to be missing something. I´m running the app with ADB using WiFi. The app crashes with and with AND without a device connected.
PD: This code is just for testing the connection.
Thanks for the help.

Aromboy, 2019-01-23:
> [Please follow the example ](https://github.com/felHR85/UsbSerial/blob/master/example/src/main/java/com/felhr/serialportexample/UsbService.java). Where are UsbDevice and UsbDeviceConnection instantiated?

I´m using the example you passed in my UsbService java class , i just want to check if the port is connected or supported.

I added this into my UsbService java class:

    public void conexion(){
        UsbDevice mc100 = null;
        UsbDeviceConnection usbconnection = null;
        UsbSerialDevice serial  = UsbSerialDevice.createUsbSerialDevice(mc100,usbconnection);
        boolean support = serial.isSupported(mc100);
        if (support == true){
            Toast.makeText(this,"sopoortado",Toast.LENGTH_SHORT).show();
        } else {
            Toast.makeText(this,"no sopoortado",Toast.LENGTH_SHORT).show();

        }
    }
And in the fragment I wrote:

    public class EstadoFragment extends Fragment {

    private Button boton;

    @Nullable
    @Override
    public View onCreateView(@NonNull LayoutInflater inflater, @Nullable ViewGroup container, @Nullable Bundle savedInstanceState) {
        View view = inflater.inflate(R.layout.fragment_estado,container,false);
        boton= view.findViewById(R.id.connect);

        boton.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {

                USBservice srv = new USBservice();
                srv.conexion();
            }
        });
        return view;

    }
}

Now it crashes when i click the button.

felHR85, 2019-01-23:
Why do you pass null values?
```java
UsbDevice mc100 = null;
UsbDeviceConnection usbconnection = null;
UsbSerialDevice serial  = UsbSerialDevice.createUsbSerialDevice(mc100,usbconnection);
```

Aromboy, 2019-01-23:
> Why do you pass null values to createUsbSerialDevice?
> 
> ```java
> UsbDevice mc100 = null;
> UsbDeviceConnection usbconnection = null;
> UsbSerialDevice serial  = UsbSerialDevice.createUsbSerialDevice(mc100,usbconnection);
> ```
Because it sends me an error in createUsbSerialDevice line, "Variable might not have been initialized"

Aromboy, 2019-01-23:
I saw it, even if i copy-paste it, then how do I check the connected device in other fragment. Again sorry for my ignorance

felHR85, 2019-01-23:
That's more generic Android than UsbSerial related. Check out for posts like Activities-Fragment-Services communication. https://hackernoon.com/8-ways-to-communicate-between-fragment-and-activity-in-android-apps-235b60005d04

Aromboy, 2019-01-23:
Maybe I´m not explaining well. I know that, but i don´t know how to call for a check of devices connected in another place that is not UsbService.java ... I will look some examples maybe i found something usefull

felHR85, 2019-01-23:
I wouldn't call for check devices elsewhere. try to call functions from UsbService from the Activity making them public because UsbService is bind to the Activity. look at findSerialPortDevice() for example

Aromboy, 2019-01-23:
I called findSerialPortDevice() from UsbService before all of this, and it also crashed . I don´t know what its bad

Aromboy, 2019-01-25:
I have been looking some examples. I managed to make the app to not crash, but still not working, The actual code is this: 

    public class EstadoFragment extends Fragment {
    
    private EditText editText;
    private Button boton;
    private USBservice usbService;
    private Handler mHandler;
    private TextView display;

    private final BroadcastReceiver mUsbReceiver = new BroadcastReceiver() {
        @Override
        public void onReceive(Context context, Intent intent) {
            switch (intent.getAction()) {
                case USBservice.ACTION_USB_PERMISSION_GRANTED: // USB PERMISSION GRANTED
                    Toast.makeText(context, "USB Ready", Toast.LENGTH_SHORT).show();
                    break;
                case USBservice.ACTION_USB_PERMISSION_NOT_GRANTED: // USB PERMISSION NOT GRANTED
                    Toast.makeText(context, "USB Permission not granted", Toast.LENGTH_SHORT).show();
                    break;
                case USBservice.ACTION_NO_USB: // NO USB CONNECTED
                    Toast.makeText(context, "No USB connected", Toast.LENGTH_SHORT).show();
                    break;
                case USBservice.ACTION_USB_DISCONNECTED: // USB DISCONNECTED
                    Toast.makeText(context, "USB disconnected", Toast.LENGTH_SHORT).show();
                    break;
                case USBservice.ACTION_USB_NOT_SUPPORTED: // USB NOT SUPPORTED
                    Toast.makeText(context, "USB device not supported", Toast.LENGTH_SHORT).show();
                    break;
            }
        }
    };

    private final ServiceConnection usbConnection = new ServiceConnection() {
        @Override
        public void onServiceConnected(ComponentName arg0, IBinder arg1) {
            usbService = ((USBservice.UsbBinder) arg1).getService();
            usbService.setHandler(mHandler);
        }

        @Override
        public void onServiceDisconnected(ComponentName arg0) {
            usbService = null;
        }
    };

    @Nullable
    @Override
    public View onCreateView(@NonNull LayoutInflater inflater, @Nullable ViewGroup container, @Nullable Bundle savedInstanceState) {
        View view = inflater.inflate(R.layout.fragment_estado,container,false);

        editText=view.findViewById(R.id.edittext);
        boton= view.findViewById(R.id.connect);
        display = view.findViewById(R.id.state);
        mHandler = new MyHandler(this   );

        boton.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {
                if (!editText.getText().toString().equals("")) {
                    String data = editText.getText().toString();
                    if (usbService != null) { // if UsbService was correctly binded, Send data
                        display.append(data);
                        usbService.write(data.getBytes());
                    }
                }
            }
        });
        return view;

[... 302 lines cut ...]

                    // Everything went as expected. Send an intent to MainActivity
                    Intent intent = new Intent(ACTION_USB_READY);
                    context.sendBroadcast(intent);
                } else {
                    // Serial port could not be opened, maybe an I/O error or if CDC driver was chosen, it does not really fit
                    // Send an Intent to Main Activity
                    if (serialPort instanceof CDCSerialDevice) {
                        Intent intent = new Intent(ACTION_CDC_DRIVER_NOT_WORKING);
                        context.sendBroadcast(intent);
                    } else {
                        Intent intent = new Intent(ACTION_USB_DEVICE_NOT_WORKING);
                        context.sendBroadcast(intent);
                    }
                }
            } else {
                // No driver for given device, even generic CDC driver could not be loaded
                Intent intent = new Intent(ACTION_USB_NOT_SUPPORTED);
                context.sendBroadcast(intent);
            }
        }
    }
}

It is pretty much generic I think , but in one of your examples it works fine and detect the conected devices, but in my app it does nothing when connecting something. Thanks for your patience and time.

closed 2019-03-09 by felHR85