felHR85/UsbSerial #203: Example Multipleports

MaDoeb, 2019-01-30:
I tested the example multipleports with one USB device (CDC) wich has two serial ports. It works great. But when I disconnect the USB device the  system crahes with the following exception:

`    --------- beginning of crash
01-16 10:58:02.004 2972-2972/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.felhr.examplemultipleports, PID: 2972
    java.lang.RuntimeException: Error receiving broadcast Intent { act=android.hardware.usb.action.USB_DEVICE_DETACHED flg=0x10 (has extras) } in com.felhr.examplemultipleports.UsbService$1@19a0290
        at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:891)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:148)
        at android.app.ActivityThread.main(ActivityThread.java:5417)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
     Caused by: java.lang.IllegalStateException: Stream contains more than one element
        at com.annimon.stream.Stream.findSingle(Stream.java:2005)
        at com.felhr.usbserial.SerialPortBuilder.disconnectDevice(SerialPortBuilder.java:135)
        at com.felhr.examplemultipleports.UsbService$1.onReceive(UsbService.java:85)
        at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:881)
        at android.os.Handler.handleCallback(Handler.java:739) 
        at android.os.Handler.dispatchMessage(Handler.java:95) 
        at android.os.Looper.loop(Looper.java:148) 
        at android.app.ActivityThread.main(ActivityThread.java:5417) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616) 
`

Source code in SerialPortBuilder.java where the exception occurs:

`public boolean disconnectDevice(UsbDevice usbDevice){
        Optional<UsbSerialDevice> optionalDevice = Stream.of(serialDevices)
                .filter(p -> usbDevice.getDeviceId() == p.getDeviceId())
                .findSingle();
`

The function **.findSingle();** seems to produce the error.

Has anyone an idee how to solve this?

Best regards Marc

felHR85, 2019-01-30:
@MaDoeb Could you send me more information about the device you are connecting? You can get with lsusb -l in Linux/OSX or using my free app DroidTerm using the USB serial viewer option

felHR85, 2019-01-30:
@MaDoeb Could you try with multiple_ports_fix branch?

felHR85, 2019-01-31:
@MaDoeb  What device are you connecting? I dont have any Usb device with two serial ports

MaDoeb, 2019-01-31:
@felHR85 The device is a digital radio from Motorola. Beforde disconnection, both ports ar working.