felHR85/UsbSerial #213: How to serial read until char?

vinzzz34, 2019-02-21:
Hi all, i need help to read commands from my arduino,
How to do a serial read until \n with this library,
I have tested protocole buffer withour sucess
` private UsbSerialInterface.UsbReadCallback mCallback = new UsbSerialInterface.UsbReadCallback() {
        @Override
        public void onReceivedData(byte[] arg0) {
            ProtocolBuffer buffer = new ProtocolBuffer(ProtocolBuffer.TEXT); //Also Binary
            buffer.setDelimiter("\r\n");
            buffer.appendData(arg0);
            while(buffer.hasMoreCommands()){
                String data = buffer.nextTextCommand();
                // Do your thing with textCommand
                try {
                    data = new String(arg0, "UTF-8");
                    if (mHandler != null)
                        mHandler.obtainMessage(MESSAGE_FROM_SERIAL_PORT, data).sendToTarget();
                } catch (UnsupportedEncodingException e) {
                    e.printStackTrace();
                }
            }

        }
    };`

felHR85, 2019-02-21:
@vinzzz34 Sorry. There is a bug in the documentation. ProtocolBuffer should be created outside onReceivedData

vinzzz34, 2019-02-21:
> 
> 
> @vinzzz34 Sorry. There is a bug in the documentation. ProtocolBuffer should be created outside onReceivedData

Thank you!

closed 2019-02-21 by vinzzz34