felHR85/UsbSerial #214: CDCSerialDevice Problems with "CDC_SET_CONTROL_LINE_STATE"

chrishamant, 2019-02-24:
Thanks for this awesome library!

I have an Adafruit Trinket M0 that I have streaming data over serial... 
Was banging my head against wall trying to get data from the both async and sync read()s following examples in repo. All the methods I tried never resulted in getting data from the serial connection...
Long story short I determined it wasn't a device/product ID issue (as it shows up as `bInterfaceClass        10 CDC Data` in `lsusb -v`) and the code successfully chooses the `CDCSerialDevice` implementation based on this.

Throwing things at a wall I reverted this commit:
https://github.com/felHR85/UsbSerial/pull/198/commits/e41b04b906d45c457d833cac601b54a99247e514
and lo my data started streaming in!
I think this makes sense as `lsusb -v` states:
```
CDC ACM:
        bmCapabilities       0x06
          sends break
          line coding and serial state
```
Don't know what drove this change (different issue for a different device? didn't see anything linked but I also am not terribly versed in debugging USB problems/implementations) - but I had to revert it for my needs... (didn't see a way to poke at it after the fact or change behavior via constructor)

Thanks again!

felHR85, 2019-02-24:
Hi @chrishamant 
I got some reports that it wasn't working properly with that line but frankly it has never been clear to me. This adds more light. I recently got a CDC device so I will test it again and revert it if necessary.
Could you put here the whole output of the lsusb for that particular device?

Thanks!

chrishamant, 2019-02-24:
```Bus 001 Device 028: ID 239a:801e  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x239a 
  idProduct          0x801e 
  bcdDevice            1.00
  iManufacturer           1 Adafruit
  iProduct                2 Trinket M0
  iSerial                 3 05BE7C335150464A39202020FF122A16
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x004b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       0 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      0 
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC ACM:
        bmCapabilities       0x06
          sends break
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      CDC Call Management:
        bmCapabilities       0x01
          call management
        bDataInterface          1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              16
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

closed 2019-05-01 by felHR85