felHR85/UsbSerial #215: Data Not Receiving in 9600 BaudRate

amarpulli544, 2019-02-26:
Hi EveryOne,

I'm using this library for read the data from variant baud rates like 9600,2400 within my android application.
When i was read the data from other source with the baud rate 2400 works fine. But not working 9600 baud rate. 
Why I'm not able to read data with the baud rate 9600.
How to resolve this issue. Please guide me.
My Device Android OS 5.1.1.

Warm Regards,
Amar.

makarna35, 2019-02-26:
Hi;
What you see if use 9600 baudrate?

amarpulli544, 2019-02-27:
Hi,
I didn't see anything while using 9600 baud rate. Because the device had connect to OTG. So i couldn't see log files. [Here](https://github.com/felHR85/UsbSerial) they had mentioned android 5.1.1 having a bug.

eclectice, 2019-02-27:
If you are debugging your Android device using Android Studio

Since USB ADB debugging won't work over OTG cable, you can try TCPIP (Wifi
or Ethernet) ADB debugging.

   1. On the Android device, install any Terminal Emulator available from
   Google Play Store (you can try *Terminal Emulator by Jack Palevich*
   https://play.google.com/store/apps/details?id=jackpal.androidterm).
   2. Connect to a Wifi or Ethernet network (if your Android device have
   Ethernet capability)
   3. Get the current IP address of the device (let say for example
   *10.12.25.202*).
   4. Make sure your laptop with Android Studio is also connected to the
   same Wifi or Ethernet network.
   5. Open a terminal session with Terminal Emulator and type the following
   commands line by line

*susetprop service.adb.tcp.port 5555stop adbdstart adbdnetstat | grep 5555*

   1. You should at least see the device is listening at 127.0.0.1:5555
   over tcp port 5555
   2. Now, open a *Command prompt* from your laptop with Android Studio IDE
   is still running
   3. Check if you have a valid ANDROID_HOME environment variable by typing
   at the command prompt: *echo %ANDROID_HOME%*
   4. if it returns a valid path to the Android SDK, then type at the
   command prompt: *cd %ANDROID_HOME%\platform-tools*
   5. Then type at the command prompt: *adb connect 10.12.25.202* or
   whatever the IP address of your Android device.
   6. It should return something like below:

** daemon not running; starting now at tcp:5037* daemon started
successfullyconnected to 10.12.25.202:5555 <http://10.12.25.202:5555>*

Now, you should be able to see logcats from your Android Studio IDE and
perform any adb-related Gradle tasks.

amarpulli544, 2019-02-27:
@eclectice Thanks for suggestion.
I will check and get back.

amarpulli544, 2019-02-27:
@eclectice ,
Getting error while using terminal emulator in android device.
[image].
Now what i have to do?

eclectice, 2019-02-27:
Follow the next steps already mentioned:

   1. You should at least see the device is listening at 127.0.0.1:5555
   over tcp port 5555 (done)
   2. Now, open a Command prompt from your laptop with Android Studio IDE
   is still running
   3. Check if you have a valid ANDROID_HOME environment variable by typing
   at the command prompt: echo %ANDROID_HOME%
   4. if it returns a valid path to the Android SDK, then type at the
   command prompt: cd %ANDROID_HOME%\platform-tools
   5. Then type at the command prompt: adb connect 10.12.25.202 or whatever
   the IP address of your Android device.

It should return something like below:

* daemon not running; starting now at tcp:5037
* daemon started successfully
connected to 10.12.25.202:5555

Note example: %ANDROID_HOME%\platform-tools =
C:\Users\YourUserName\AppData\Local\Android\Sdk\platform-tools

amarpulli544, 2019-02-27:
@eclectice ,
Here my android device showing 0.0.0.0:5555 ip address  in terminal emulator.
Is it fine with that.

eclectice, 2019-02-28:
0.0.0.0: 5555 or 127.0.0.1: 5555 is ok as long as it is listening at tcp
port 5555.. From your feedback, the server is now running well at port 5555
from Android device side. So now, you need to connectto it from PC side
with "adb connect <Android device IP address>". Make sure Android device
and PC are both on the same IP network.

amarpulli544, 2019-02-28:
@eclectice , Thanks for helping,
We will check and get back to with results soon!!

amarpulli544, 2019-03-06:
@eclectice ,Yes Everything work fine while using terminal emulator.
Here i am using usb to rs232 converter for getting data to application. I'm not getting any errors while receiving the from usb. I think issue with receiving the data from usb. Becoz we need to handle the data. Can you please guide me how to handle the data while receiving the data from usb.
Warm Regards,
Amar.