felHR85/UsbSerial #221: Failed to resolve: com.github.felHR85:UsbSerial:4.5.2

lindapang, 2019-03-18:
Gradle

Add the jitpack repo to your your project's build.gradle at the end of repositories

/build.gradle

allprojects {
repositories {
jcenter()
maven { url "https://jitpack.io" }
}
}

Then add the dependency to your module's build.gradle:

/app/build.gradle

compile 'com.github.felHR85:UsbSerial:4.5.2'

felHR85, 2019-03-18:
@lindapang implementation 'com.github.felHR85:UsbSerial:6.0.4'

lindapang, 2019-03-18:
Why i get this error when implementation com.github.felHR85:UsbSerial:4.5.2 ？

lindapang, 2019-03-18:
@felHR85  Why i get this error when implementation com.github.felHR85:UsbSerial:4.5.2 ？

felHR85, 2019-03-18:
@lindapang What error?

lindapang, 2019-03-18:
@felHR85 Failed to resolve: com.github.felHR85:UsbSerial:4.5.2

felHR85, 2019-03-18:
@lindapang Does it work with implementation 'com.github.felHR85:UsbSerial:6.0.4' ?

onitica, 2019-03-19:
Did you remove 4.5.2 from Jitpack? Why?

felHR85, 2019-03-19:
@onitica I didnt remove it personally. It looks like Jitpack remove old releases.

onitica, 2019-03-19:
Can you re-add it? I had releases of a demo app to clients that depended on that which are now all broken. If old releases get removed like that, I definitely can't use this library.

felHR85, 2019-03-19:
@onitica  Can't you recompile it with 6.0.4? 4.5.2 is like two years old and fixes some known issues

felHR85, 2019-03-19:
@onitica Anyways it is a big flaw if that's Jitpack default behavior. I will try to contact the team.

onitica, 2019-03-19:
I had this distributed as part of a source code example to clients and now every single on of their builds are broken. I don't think it is a good policy to ever remove old artifacts, but I didn't know that was a jitpack thing (and now I know to never use jitpack if that is true). It seems to work with `6.0.4`. I do appreciate the work you've done on this library. I actually have moved away and written our own USB implementation for our next release, but this was a great stepping stone for us to get there and get the early builds out to clients.

felHR85, 2019-03-19:
@onitica That sucks 😓. Again sorry for the problems caused.

Knoxie, 2019-03-21:
It's not good policy to release artifacts that are dependent on being accessed from the web to a client, especially something you don't maintain. You should have downloaded the source/jar and imported directly @onitica

onitica, 2019-03-21:
@Knoxie I'm not disagreeing with you, but Android Studio isn't the best with imported JAR files. Lots of build/caching issues, and also I then need to pull in any transitive JARs. The plan was to always get rid of this artifact since what was out there was a beta build they couldn't go live with anyways.

aganrian, 2019-03-22:
mee too. i cant resolve 4.5.2 can u re upload again.

abdoul-Karim, 2019-05-31:
Hello can't resolve 6.0.6

closed 2019-07-11 by felHR85

zp0105, 2023-01-20:
Failed to resolve: com.github.felHR85:UsbSerial:6.0.4   i am also facing this issue