felHR85/UsbSerial #223: Data receiving problem in example asyncronhous

augustopeterle, 2019-03-22:
I'm new in Android developing. I'm using this library to collect data from a GPS with FTDI chip.
I have a problem, when a I try to manipulate the received data in the Handler. The GPS sends data very fast (1ms) and some times in my screen it doesn't show the "\n". Why this occur ? The UART is faster than the inside Handler operation ? Before finishing the Handler instructions is there already a new data obtained from USB?

Regards.

 ` @Override
        public void handleMessage(Message msg) {
            switch (msg.what) {
                case UsbService.MESSAGE_FROM_SERIAL_PORT:
                    String data = (String) msg.obj;
                    String[] temp = data.split(",");

                    if(temp.length>3) {
                        if (temp[0].contains("GPGSV")) {
                            mActivity.get().display.append(data);
                            mActivity.get().display.append("\n");
                        }

                        if (temp[0].contains("GGA")) {
                            mActivity.get().display.append(data);
                            mActivity.get().display.append("\n");
                        }
                    }
                    break;
                case UsbService.CTS_CHANGE:
                    Toast.makeText(mActivity.get(), "CTS_CHANGE",Toast.LENGTH_LONG).show();
                    break;
                case UsbService.DSR_CHANGE:
                    Toast.makeText(mActivity.get(), "DSR_CHANGE",Toast.LENGTH_LONG).show();
                    break;
            }`

kahpooi, 2019-06-03:
Hi, I'm new in Android and trying to use this USBserial library to read data from RS232. It is working. Then, I wanted to split the string into array using split(" ") and put them into respective textviews. However, the app just closed when 'USB ready' toast was shown on the screen. Below is my code, is there any problem on the code?

public void handleMessage(Message msg) {
            switch (msg.what) {
                case UsbService.MESSAGE_FROM_SERIAL_PORT:
                    String data = (String) msg.obj;
                    mActivity.get().display.append(data);

                    String[] values = data.split(" ");
                    mActivity.get().voltage.setText(values[0]);
                    mActivity.get().current.setText(values[1]);
                    mActivity.get().power.setText(values[2]);
                    mActivity.get().temperature.setText(values[3]);
                    mActivity.get().wh.setText(values[4]);
                    mActivity.get().battvolt.setText(values[5]);
                    mActivity.get().blowrate.setText(values[6]);

                    break;

                case UsbService.CTS_CHANGE:
                    Toast.makeText(mActivity.get(), "CTS_CHANGE",Toast.LENGTH_LONG).show();
                    break;

                case UsbService.DSR_CHANGE:
                    Toast.makeText(mActivity.get(), "DSR_CHANGE",Toast.LENGTH_LONG).show();
                    break;
            }

AblertEinstein, 2021-01-19:
I have the problem, too. I think this is because the buffer size is larger than the size of the data received each time from the Usb Device. And I am trying to solve the problem. Do you have some good methods?