felHR85/UsbSerial #227: CDC Device not writing and reading data

dsuresh-ap, 2019-03-28:
We just updated our code to 6.0.4 and now we are unable to read or write data from a CDCSerialDevice. Reverting back to 5.0.1 works but none of the 6.0.x versions work. The logs show that data is being pushed to the write buffer.

It looks like the `WorkerThread` is getting stuck on `UsbRequest request = connection.requestWait();` Once the device is closed the read callback gets called. 

Might be related to https://github.com/felHR85/UsbSerial/issues/214.

felHR85, 2019-03-28:
@dsuresh-ap  Thank you for this report. I will check it out asap

Scotti6666, 2019-04-01:
I'm having the same issue with 6.0.4 and fixed it by changing CDC_CONTROL_LINE_OFF to _ON in CDCSerialDevice.java:openCDC().

        // Default Setup
        setControlCommand(CDC_SET_LINE_CODING, 0, getInitialLineCoding());
        setControlCommand(CDC_SET_CONTROL_LINE_STATE, CDC_CONTROL_LINE_ON, null);

felHR85, 2019-04-01:
@Scotti6666 @dsuresh-ap Thank you guys. I will release this weekend with this change.

felHR85, 2019-04-06:
@dsuresh-ap @Scotti6666 I just released 6.0.5 with that line changed. It would be great if you could give it a try. Thanks!

Scotti6666, 2019-04-06:
I pulled and checked out 6.0.5 and it's working. Thanks!

dsuresh-ap, 2019-04-10:
@felHR85 Looks like the fix works and I am able to communicate properly. Thank you!

closed 2019-04-15 by felHR85