felHR85/UsbSerial #233: steveohara/j2mod integration

44simon, 2019-04-14:
Hello,
thanks for such a great library. I have found steveohara/j2mod thing and I want to run a serial RTU master with your library. I have sucesfully ran an example with the FTDI device - it is detected and works just fine. Do you happen to have some examples how to do this using your library and steveohara/j2mod?

Thank you so much in advance,
Simon

felHR85, 2019-04-15:
Hi @44simon 
Actually I tried to integrate j2mod with UsbSerial but I became distracted. Sadly there are no examples as far as I know.
Maybe it's time to fork and maintain an Android version with UsbSerial and gradle with the shared code with J2mod. I will think about it

44simon, 2019-04-15:
I can help you with testing and documenting.

felHR85, 2019-04-15:
@44simon That would be great! I will keep updated this issue

felHR85, 2019-04-19:
@44simon 
I just created a repo with a really rushed first version of an integration between j2mod and UsbSerial
https://github.com/felHR85/AndroidModbus
It contains two modules. The library itself and an example app. Example app managed to send some data using this:
```java
modbusSerialMaster.writeCoil(1, true);
```
 but I guess this waits for a response and it is blocking the main thread. My modbus knowledge is basic so it would be nice if you could test proper Modbus communications with this.

closed 2019-07-11 by felHR85