felHR85/UsbSerial #236: Can anyone guess why UsbSerialDevice is null?

kotran88, 2019-04-22:
Hello :) 
I have cp2102 and usb -c , esp8266 to send data to phone. 

my custom pcb is wired like
cp2102's D+ and D- is connected to usb-c 's D+ and D-
and cp2102 ' rx tx is connected to mcu's tx,rx

I tested code below to node mcu which has cp2102 and work quite well(UsbSerialDevice is not null and later, open well)

           ```
 connection = manager.openDevice(device);
            Log.e("dialog","device is : "+device.toString());
        Log.e("dialog","connection is : "+connection);

        UsbSerialDevice serialPort = UsbSerialDevice.createUsbSerialDevice(device, connection);
```

according to Log above. 
UsbDevice[mName=/dev/bus/usb/001/002,mVendorId=4554,mProductId=529,mClass=0,mSubclass=0,mProtocol=0,mManufacturerName=Silicon Labs,mProductName=Verifone USB to Modem,mVersion=1.00,mSerialNumber=0001,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=50,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=Verifone USB to Modem,mClass=255,mSubclass=0,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=2,mMaxPacketSize=64,mInterval=0]
    UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=64,mInterval=0]]]]

and
connection is : android.hardware.usb.UsbDeviceConnection@d695098

but 

UsbSerialDevice serialPort = UsbSerialDevice.createUsbSerialDevice(device, connection);

here comes error which serialPort is null...so I can't open it. 

can anyone guess why UsbSerialDevice is null even though device and connection is not null

felHR85, 2019-04-22:
What are your VID/PID? That methods only return null when the device is not supported

felHR85, 2019-04-22:
If problem still persists though. You could use
```java
UsbSerialDevice serial = UsbSerialDevice.createUsbSerialDevice(UsbSerialDevice.CP210x, device, connection,0);
```

kotran88, 2019-04-23:
Hello :) I'm using cp2102 bought from digikey...
as I said Device id is same as node mcu but product id is diffeerent...and name is different...
node mcu's usbdevice name is cp2102 usb to uart but my custom pcb 's usbdevice name is Verifone USB to Modem...it may be the reason library think my pcb is not supported...?

felHR85, 2019-04-23:
@kotran88 
Probably, Try to open it as I told you before :)

closed 2019-04-30 by felHR85