felHR85/UsbSerial #238: Write for async write function finishes

efreibe, 2019-04-24:
Hello,

I'm trying to open the usb port, send data and close it. But I realized that not all the data
is not transferred to destination.
I suspect that while writing I send a close command and my data is truncated.
How can I listen to know when all buffer was totally written to the port?

Thank you

felHR85, 2019-04-25:
Hi @efreibe 
[I would use the synchronous api  in your case](https://github.com/felHR85/UsbSerial/wiki/5.-Synchronous-api)

efreibe, 2019-04-26:
Yes, I'm doing that, I've created a thread and I'm making calls to syncWrite

I simply created the issue because it will be helpful for other people making async requests 
to get a notification when the write operation is done.

I'm using flutter and the easy way to send a success message back to the channel is waiting for a notification from write async method. But I managed to do it in other way.

You can close it, it's not a blocking issue.

felHR85, 2019-04-26:
@efreibe  Like a JS promise with "then" added at the end to handle the result of the operation?
I guess it could be added because it is a nice idea.

efreibe, 2019-04-27:
Good to hear that. It will give more control to the write process.