felHR85/UsbSerial #239: CH340, Cannot read any bytes

WangLiang85, 2019-04-25:
I am trying convert esptool.py to android.
For serial communication, I am using this library.

From https://github.com/espressif/esptool/wiki/Serial-Protocol, 
We can send command bytes to esp8299 while bootloader mode.

My project is working well with esp8266/cp2102, esp8266/ftdi.
But not working with esp8266/ch340.

**Strange:**
With esp8266/ch340, read/write is working well in normal mode(not bootloader mode).

but, at bootloader mode, I can read only [Boot-Rom-log](https://github.com/espressif/esptool/wiki/ESP8266-Boot-ROM-Log)

For read boot-rom-log, I need buadrate 74880.
So I merged https://github.com/kai-morich/usb-serial-for-android code to [this project](https://github.com/felHR85/UsbSerial).

With esp8266/ch340, if I make it bootloader mode, the I can read [Boot-Rom-log](https://github.com/espressif/esptool/wiki/ESP8266-Boot-ROM-Log).

But after, I cannot read any bites from that.
From https://github.com/espressif/esptool/wiki/Serial-Protocol, if I write **sync** bytes, then the esp8266 send response bytes.

**More Strange**
I can read Response bytes from esp8266/ch340 with disconnect it from android phone.
At moment while disconnecting, the response bytes are arrived.

Or, I can read Response bytes while reboot esp8266/ch340 via RST button.

More Details are mentioned in [StackOverflow](https://stackoverflow.com/questions/55672193).

felHR85, 2019-04-25:
@WangLiang85 Maybe you should ask in the other library issue tracker. This is only for issues related with this library

WangLiang85, 2019-04-26:
As I mentioned, I am using this library.