felHR85/UsbSerial #241: FTDI device syncRead not return data

t2t-sonbui, 2019-04-26:
I run _examplesync_ with FT232RL chip, but can not get anything received.  syncRead(byte[] buffer, int timeout) with _timeout_ value is 0 will make the _ReadThread_ unable to stop.

arabine, 2019-06-19:
Same here. Must have a problem in the current implementation...

atholbro, 2019-07-03:
It's working with Release 6.0.3, anything newer and syncRead returns 0. This is with a 1000ms timeout.

felHR85, 2019-07-03:
@t2t-sonbui @arabine @atholbro 
You are totally right. It is confirmed that a bug was introduced. This bug only affects FTDI devices using synchronous api or Streams.
I will release a new version this weekend. In the meantime I recommend using the asynchronous api.
Sorry guys for the inconvenience.

atholbro, 2019-07-03:
I'm just using 6.03 for now.

I'm also having an issue with larger reads failing. It seems to read about 230 bytes then cuts off. Still looking into it as it might be something on my end.

felHR85, 2019-07-05:
@t2t-sonbui @arabine @atholbro 
Hey guys. Could you give this branch **ftdi_sync_fix** a try?
Thank you very much

felHR85, 2019-07-05:
> I'm just using 6.03 for now.
> 
> I'm also having an issue with larger reads failing. It seems to read about 230 bytes then cuts off. Still looking into it as it might be something on my end.
Maybe you could increment the buffer?

arabine, 2019-07-10:
Ok I will try it.

felHR85, 2019-07-11:
Solved in 6.1.0 release

closed 2019-07-11 by felHR85