felHR85/UsbSerial #244: Just curious as to why you don't do release notes?

Knoxie, 2019-05-01:
Lots of ways to automate it, Makes people more likely to update knowing what changed, and lets people know if breaking changes may happen for them.

felHR85, 2019-05-01:
Since 6.0.1 I added a CHANGELOG. But further ideas of improving releases are welcomed.

Knoxie, 2019-05-01:
[image]

Just putting the change log stuff in there would be nice, instead of having to go into the changelog, but changelog is nice.  A list of all the closed 'issues' in the list would be nice.

Features:
issue 10: blah blah blah
issue 12: blah blah blah
issue 14: blah blah blah

Bug Fixes:
issue 13: blah blah
issue 15: blah blah

Just an idea, appreciate all the work you do on this!

Knoxie, 2019-05-01:
[image]

I'm just basing this idea off other repos: https://github.com/home-assistant/home-assistant/releases

closed 2019-05-01 by Knoxie

felHR85, 2019-05-01:
Those are great ideas. I will keep them in mind for the next release.