felHR85/UsbSerial #247: Problem with Multiple Serial ports and FT2232H

W1il, 2019-05-09:
Hi, I have been testing the example of Multiple Serial ports using a double USB-RS232 converter which internally has two FT232R and works OK, but this manufacturer also has another model of the[ FT2232H](https://www.ftdichip.com/Products/ICs/FT2232H.html) controller this is a double USB-RS232 converter all in a single circuit integrated, and it would be the ideal match for your app, but when I test with this controller it only lists a USB-RS232 port and does not double as it should be

With the double converter FT232R I have the following:
UsbPort:/dev/bus/usb/001/004
DeviceId:1004
ProductId:24577
VendorID:1027

UsbPort:/dev/bus/usb/001/003
DeviceId:1003
ProductId:24577
VendorID:1027

But with the FT2232H converter I have the following:
UsbPort:/dev/bus/usb/001/002
DeviceId:1002
ProductId:24592
VendorID:1027

As you can see the Android only recognizes a single USB-RS232 port and does not double as it should be

Do you have any suggestions on how to correct this?

felHR85, 2019-05-09:
@W1il 
Both should be work but I couldn't test it with any sample of a multiple serial port inside the same chipset.
Could you send me more information about the USB device [it can be done with a tool like this](https://www.thesycon.de/eng/usb_descriptordumper.shtml)

W1il, 2019-05-10:
@felHR85,

Here is the damp file.

[DescriptorDump_Debuger-2UART.txt](https://github.com/felHR85/UsbSerial/files/3164425/DescriptorDump_Debuger-2UART.txt)

felHR85, 2019-05-11:
Hi @W1il 
[Probably still not fixed but please try to use the multiple port example of this branch with both devices](https://github.com/felHR85/UsbSerial/tree/ftdi_multiple_interfaces)

felHR85, 2019-07-11:
@W1il Could you test it with 6.1.0 release?

closed 2019-07-11 by felHR85

W1il, 2019-08-08:
@felHR85, 
I just did the test with 6.1.0 release, just detect only a single port and not double as it should be the FT2232