felHR85/UsbSerial #248: Unity

Scott12334, 2019-05-21:
Hello,
Is there any way to make this work with unity?

felHR85, 2019-05-22:
@Scott12334 
Sure there is a way but I don't really know

Scott12334, 2019-05-24:
Ok thank you, I am doing it with Android Studio now and the installation does not seem to be working. Attached I have the gradle dependency and repository code. No errors when syncing with the rest of the project but UsbSerialDevice doesn't work. I am following this video: https://www.youtube.com/watch?v=QHa6HWTmQFs. Any ideas?
[image]
[image]

kamran1993, 2019-07-06:
yes i use this library in unity3d first : you most write code in android studio and make a jar file 
but there is some tip on it , i commit my example project in github as soon as possible.

felHR85, 2019-07-06:
@kamrankhorasani I would love to add a link to your example project.

kamrankhorasani, 2019-07-06:
@felHR85 thank you Felipe , I use your library to a large project but I decide write a plug-in and example of your library that used in unity . So when I write this plug-in send message to you to add my link to your Page .

Scott12334, 2019-07-06:
Could also send me that link?
Thanks