felHR85/UsbSerial #249: NullPoint Exception

Scott12334, 2019-05-26:
Hello,
So I tried running the code, and the USB device code does not seem to be working. It returns a null value and then says there is a NullPointException. Thank you in advance, and I really appreciate this amazing library really makes life easy. Here are some screenshots. 
[image]

felHR85, 2019-05-26:
Hi @Scott12334 
It is a NullPointerException because both device and usbConnection are null.
[Please use this Service as an example](https://github.com/felHR85/UsbSerial/blob/master/example/src/main/java/com/felhr/serialportexample/UsbService.java). findSerialPortDevice() method is used to properly handle the device variable

Scott12334, 2019-05-26:
Sorry, I don't quite understand. How would I implement findSerialPortDevice()? The service you showed didn't show that. Sorry, I am new to android and I really appreciate your help.

felHR85, 2019-05-26:
@Scott12334 
For Android beginners I recommend using the example as a starting point and make your changes there.

Scott12334, 2019-05-26:
Ok so you recommend I copy that code and make that my service?

felHR85, 2019-05-26:
@Scott12334 
And even the Activity if you are not so sure.

Scott12334, 2019-05-26:
The problem is I understand my code right now, except for that one error. Is there a simple fix using findSerialPortDevice. Thank you

felHR85, 2019-05-26:
UsbSerialDevice serial = UsbSerialDevice.createUsbSerialDevice(device, usbConnection); 
This line can't work because both arguments are null. You are declaring it when the service class is init and thats the reason why both are null (even device is set directly to null). Please refer to the example

Scott12334, 2019-05-26:
So instead of doing this code serial.read(mCallback);
            int UnityPrint = serial.read(mCallback); do this
 mHandler.obtainMessage(MESSAGE_FROM_SERIAL_PORT, data).sendToTarget();
that way i dont have to use serial.

Scott12334, 2019-05-26:
I am really sorry but I don't see a part of the example that fixes that line. The example just skips that line

Scott12334, 2019-05-27:
Sorry about that I found it

Scott12334, 2019-05-27:
Ok so I found it but now I am having trouble using it. I simply just want to have that device that is found to become private UsbDevice device; None of the vendor ID or permission requests. Thank you for your help

Scott12334, 2019-05-27:
This is what I have tried: 
[image]