felHR85/UsbSerial #259: save permission without any intent filter

kamrankhorasani, 2019-07-08:
is there any solution to save usb host permission for next use of usb device without any intent filter?

dipakbari4, 2019-07-12:
Yes, I think you should have to use SharedPreference and pass the data on next use from shared preference.

kamrankhorasani, 2019-07-13:
> Yes, I think you should have to use SharedPreference and pass the data on next use from shared preference.

i want set permission first time for some usb device and connect to them and for second time when i plug in my usb device i dont want request any permission for this usb and i dont want using any intent filter . you said save data to shared preference . so what data should i save and how to use them to prevent any permission request for usb device?

dariocavada, 2021-07-01:
> i want set permission first time for some usb device and connect to them and for second time when i plug in my usb device i dont want request any permission for this usb and i dont want using any [...]

Any advice on that ?