felHR85/UsbSerial #263: on app, I wrote with Serial.write("rrr".getBytes()), but arduino doesn't received.

kotran88, 2019-07-16:
android code is as below
I checked Serialport is opened, then I write string date to arduino

` serialPort = UsbSerialDevice.createUsbSerialDevice(UsbSerialDevice.CP210x, device, connection,0);
//        UsbSerialDevice serialPort = UsbSerialDevice.createUsbSerialDevice(device, connection);
                    if (serialPort != null) {
                        Toast.makeText(getApplicationContext(),"opened!!!",Toast.LENGTH_SHORT).show();
                        Log.e("dialog","serialport : "+serialPort.toString());

                        serialPort.setBaudRate(9600);
                        serialPort.setDataBits(UsbSerialInterface.DATA_BITS_8);
                        serialPort.setStopBits(UsbSerialInterface.STOP_BITS_1);
                        serialPort.setParity(UsbSerialInterface.PARITY_NONE);
                        serialPort.setFlowControl(UsbSerialInterface.FLOW_CONTROL_OFF);

                        serialPort.write("rrr".getBytes());

                        serialPort.read(mCallback); //

                    } else {
                        Log.e("dialog", "PORT IS NULL");
                    }`

on arduino, 
loop(){
}

I added blink to recognize arduino is received string data well. 
but not blinking at all...
what did I do wrong.

kotran88, 2019-07-16:
my git address to refer is...

https://github.com/kotran88/SerialPrinting

feder240516, 2020-04-30:
could you solve this? having same problem here

helaquiz, 2020-06-04:
You need to write in other function see the example https://github.com/felHR85/UsbSerial/tree/master/example

And i arduino you need to begin the serial port