felHR85/UsbSerial #272: Com port "proxy" for non rooted devices

podarok, 2019-08-14:
Hi
There is a great Linux like system for Android - termux.
And the only issue to convert any Android to perfect IOT thing is access to com ports
Right now
[image]

Which is a bit lame...
Only one step to get access to any usb device from programs like zigbee2mqtt , which could be compiled for Android and work well.

Are there 
- any plans to provide some sort of the proxy for user level environment to be able to access any device via USB?
- if no - is there an example code for how to approach this?

Thanks in advance

felHR85, 2019-08-14:
Hi! @podarok 
I am afraid you need a rooted device for that. This library implements Usb to serial drivers in the user space precisely because that limitation.
Using a rooted device you can access /dev/ttyXXX normally using Process class or even with C code and glue it with JNI

felHR85, 2019-08-14:
@podarok By the way it looks like you are using a X96. We've been evaluating some of them in where I work and it should be easy to have them root.

podarok, 2019-08-14:
would be cool to have it non rooted @felHR85 
Kinda opens easy way for wider community, not only for nerds