felHR85/UsbSerial #273: SerialPortBuilder crash on detaching FT2232

changwoo, 2019-08-14:
On version 6.0.1, it always crashes on detaching FT2232. Maybe because it's a dual RS232 device?

```
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.myapp.app, PID: 4812
    java.lang.RuntimeException: Error receiving broadcast Intent { act=android.hardware.usb.action.USB_DEVICE_DETACHED flg=0x1000010 (has extras) } in com.myapp.app.SerialDevice$UsbReceiver@305760d
        at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$-android_app_LoadedApk$ReceiverDispatcher$Args_52497(LoadedApk.java:1323)
        at android.app.-$Lambda$aS31cHIhRx41653CMnd4gZqshIQ.$m$7(Unknown Source:4)
        at android.app.-$Lambda$aS31cHIhRx41653CMnd4gZqshIQ.run(Unknown Source:39)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6494)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
     Caused by: java.lang.IllegalStateException: Stream contains more than one element
        at com.annimon.stream.Stream.findSingle(Stream.java:2005)
        at com.felhr.usbserial.SerialPortBuilder.disconnectDevice(SerialPortBuilder.java:139)
        at com.myapp.app.SerialDevice$UsbReceiver.onReceive(SerialDevice.java:184)
        at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$-android_app_LoadedApk$ReceiverDispatcher$Args_52497(LoadedApk.java:1313)
        at android.app.-$Lambda$aS31cHIhRx41653CMnd4gZqshIQ.$m$7(Unknown Source:4) 
        at android.app.-$Lambda$aS31cHIhRx41653CMnd4gZqshIQ.run(Unknown Source:39) 
        at android.os.Handler.handleCallback(Handler.java:790) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loop(Looper.java:164) 
        at android.app.ActivityThread.main(ActivityThread.java:6494) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807) 
```

felHR85, 2019-08-14:
@changwoo  Please update to 6.1.0 That should be fixed there

changwoo, 2019-08-14:
My bad. didn't know the recent one was 6.1.0.

closed 2019-08-14 by changwoo