felHR85/UsbSerial #275: Is the CP2105 dual-port in one chip supported?

obbardc, 2019-08-19:
I am about to use the CP2105 in a new design, with both serial ports.

The VID/PID will be detected, but I am not sure if the second port will work out of the box?

Looking at the code for the CP2102 I can't see how the second port will work?
 https://github.com/felHR85/UsbSerial/blob/master/usbserial/src/main/java/com/felhr/usbserial/CP2102SerialDevice.java

https://www.silabs.com/documents/public/data-sheets/CP2105.pdf

Thanks :-)

felHR85, 2019-08-19:
@obbardc 
Using SerialPortBuilder class it will try to open it as two CP2102.
Didnt know this particular device so some feedback would be appreciated :)

obbardc, 2019-08-19:
The hardware will be approx 1 month, I will keep this issue updated as soon as I can...
Cheers!

zcsrf, 2019-08-22:
I have used the CP2105 and it shows as a second iface.

Also as a side note, RTS toggle works fine, but the DTR doesn't not toggle, it seems that there is a difference between the behaviour of the pins of the CP2105 and the CP2102.

I have yet to fully test this, but based on the AN571 - https://www.silabs.com/documents/public/application-notes/AN571.pdf, using the VENDOR_SPECIFIC (0xFF) register one can possible change/read the GPIO state of the associated DTR pin.

obbardc, 2019-09-16:
Yes, it works correctly out-of-the-box ;-).

obbardc, 2019-09-18:
Hi @felHR85

I am trying to use the example with my own thread to read from the serial port, https://github.com/felHR85/UsbSerial/blob/master/examplemultipleports/src/main/java/com/felhr/examplemultipleports/UsbService.java#L189

and it's not very reliable. We are seeing missed messages.

If I open one port with the sync api and set a callback using serialPort_COM2.read(dataCallback_COM2); all works well.

But I cannot open two ports in this way, it never seems to read data from the second port?

```
        // open device
        connection = usbManager.openDevice(device);

        Log.info("openSerialPorts() opening COM1");
        serialPort_COM1 = UsbSerialDevice.createUsbSerialDevice(device, connection, COM1);
        if (serialPort_COM1 != null) {
            if (serialPort_COM1.open()) {
                Log.info("openSerialPorts() opened COM1");
                serialPort_COM1.setBaudRate(BAUD_RATE);
                serialPort_COM1.setDataBits(UsbSerialInterface.DATA_BITS_8);
                serialPort_COM1.setStopBits(UsbSerialInterface.STOP_BITS_1);
                serialPort_COM1.setParity(UsbSerialInterface.PARITY_NONE);
                serialPort_COM1.setFlowControl(UsbSerialInterface.FLOW_CONTROL_OFF);
                serialPort_COM1.read(dataCallback_COM1);
            }
        }

        Log.info("openSerialPorts() opening COM2");
        serialPort_COM2 = UsbSerialDevice.createUsbSerialDevice(device, connection, COM2);
        if (serialPort_COM2 != null) {
            if (serialPort_COM2.open()) {
                Log.info("openSerialPorts() opened COM2");
                serialPort_COM2.setBaudRate(BAUD_RATE);
                serialPort_COM2.setDataBits(UsbSerialInterface.DATA_BITS_8);
                serialPort_COM2.setStopBits(UsbSerialInterface.STOP_BITS_1);
                serialPort_COM2.setParity(UsbSerialInterface.PARITY_NONE);
                serialPort_COM2.setFlowControl(UsbSerialInterface.FLOW_CONTROL_OFF);
                serialPort_COM2.read(dataCallback_COM2);
            }
        }
```

I'm not sure if you can suggest a way where I can open both ports and reliably not miss any data ?

Thanks!

obbardc, 2019-09-23:
Hi @felHR85 any ideas on how best to proceed ?

felHR85, 2019-09-23:
Hi @obbardc 
I will check it out as soon as possible.

obbardc, 2019-10-07:
Hi @felHR85 
I think I have figured it out and have commented my thoughts here: https://github.com/felHR85/UsbSerial/issues/280

Cheers!