felHR85/UsbSerial #280: Data arrives out of order

shakram02, 2019-09-04:
Hello, thank you for making this library available, I'm having problems reading data. This is the sample code I'm using on Arduino

```Arduino
void write_serial_bytes(int count) {
  for (int i = 0; i < count ; i++) {
    Serial.write(random('A', 'Z'));
    if (i != count - 1)
    {
      Serial.write(",") ;
    }
  }
  Serial.write("&");
}
```
### Problem
The data doesn't arrive in the order it's written by. This format sends a character followed by a comma but sometimes I either find 2 consecutive characters or two commas which means that data arrives out of order

I made a python program to verify if the problem is from Arduino but it didn't seem to be the case. Maybe that happens because I issue multiple writes?, also sometimes I receive empty data events (when I used the `async` api). It seems like a race problem but I'm not sure

I tried using both sync and async APIs.

soreau, 2019-09-07:
I am having a similar issue. I send 3 bytes with delimiter, always the same, as a test. On android, I receive the data but every once in awhile, data[318] == data[319]. As you can see in the comment, there's no way to tell if the data is duplicated legitimately or erroneously, when the data is real audio data. This doesn't happen on linux using python serial.Serial.
```
sprintf(buf, "%c%c\n", 0x80, 0x90);
Serial.print(buf);
```
```

UsbSerialInterface.UsbReadCallback mCallback = (byte[] data) -> {
    int length = data.length;
    byte[] buf = new byte[length];
    boolean split = false;
    int size = 0, i;

    if (length == 0) {
        // This does happen
        return;
    }

    if (last_end_byte != '\n' && data[0] != '\n') {
        buf[size++] = last_end_byte;
    }

    last_end_byte = data[length - 1];

    for (i = 0; i < length; i++) {
        if (data[i] == '\n')
            continue;

        // HACK: Work around bug where we get duplicate data
        // However, this can fail with real data because
        // there's no way to tell if the data is mistakenly
        // duplicate or actually duplicated on the device,
        // in which case we shouldn't continue here.
        if (i == 319 && data[318] == data[319])
                continue;

        buf[size] = data[i];

        if ((size % 2) == 1)
            audioTrack.write(buf, 0, 2);

        size++;

        if (size > 1)
            size = 0;
    }
};
```

soreau, 2019-09-07:
I've discovered it also happens sometimes where data[319] == data[320] with this modified check.
```
if ((i >= 2) && (data[i - 1] == data[i]) && ((data[i - 2] != '\n') || (((i + 1) < length) && (data[i + 1] != '\n')))) {
    Log.d(TAG, "continuing: " + i);
    continue;
}
```

mtinnes, 2019-09-17:
Same issue communicating with Arduino Uno (Possibly a CDC driver). Reads contain out of order and/or duplicated characters.

shakram02, 2019-09-17:
@felHR85 any clue on where a fix can be done? that'll be pretty helpful

mtinnes, 2019-09-23:
I tracked it down to the read() method in the SerialInputStream.java. I don't think data should be added to the buffer if syncRead() returns 0;

```
    @Override
    public int read()
    {
        int value = checkFromBuffer();
        if(value >= 0)
            return value;

        int ret = device.syncRead(buffer, timeout);
        if(ret > 0) {
            bufferSize = ret;
            return buffer[pointer++] & 0xff;
        }else {
            return -1;
        }
    }

```

shakram02, 2019-09-24:
I sent felHR85 an email, he said that he'll check it in the weekend

obbardc, 2019-10-07:
Hi,

My project has a lot of data flowing over two serial ports, and I think my bug may be related.
The serial port has messages send like ":test test test;", using the colons as a start & stop character.
What I see received by the library is "::test test test;" in about 50% of all messages.

I've found using the asynchronous functions `UsbSerialDevice.read()` with a callback works perfectly fine, all of the data seems to be present. This would be fine normally, but the asynchronous functions can only be used with one port at a time.

So opening the serial port using the builder `SerialPortBuilder.openSerialPorts()` and the synchronous functions the following is true:-

- using `UsbSerialDevice.getInputStream()` and then reading each character in a separate thread per serial port (like the examplemultipleports using `read()`) the data arrives mangled like above.
- using `UsbSerialDevice.syncRead(buf, 256)` works okay, there is no mangled characters.

So maybe `SerialInputStream` is not reading from the correctly?
I think I have solved my issue using `syncRead`.

subvertallchris, 2020-12-29:
Any chance someone has a forked version of this library that fixes this?

subvertallchris, 2020-12-29:
Any chance this got fixed and the issue wasn't closed?