felHR85/UsbSerial #284: okio.buffer not found

mbhassan, 2019-10-06:
Having an issue when calling createUsbSerialDevice function.

using v6.1.0

javadang.NoClassDefFoundError: okio.Buffer
com.felhhusbmrial.SerialBuffer$SynchronizedBuffeheine,(Sed  
alBuffehjava:97)
com.felhhusbmrial.SerialBuffeheine,(SerialBuffehjave:21)
com.felhhusbmrial.UsbSerialDevice.einit>(.1obSerialDevice.jave  
:58)
com.felhhusbmrial.FTDISerialDevice,init4TDISerialDevice.ja  
ve:110)
com.felhhusbmrial.UsbSerialDevice.createUsbSerialDevice(Usb  
Serial0evice.jave:77)
com.felhhusbmrial.UsbSerialDevice.createUsbSerialDevice(Usb  
SerialDevice.jave:53)
cornins.Runtime.calIJSMethodNative(Native Method)
comIns.Runtime.dispetchCallJSMethodNative(Runtime.javal  
209)
comIns.Runtime.calIJSMethodImpl(Runtime.jave:1095)  
comIns.Runtime.calIJSMethod(Runtime.javal 083)  
comIns.Runtime.calIJSMethod(Runtime.javal 053)  
comIns.Runtime.calIJSMethod(Runtime.javal 055)
comIns.gen.javadang.Object_vendor_119581_32_ClickListener  
ImpLonClick(Object_vendor_119581_32_ClickListenerImpl.java:  
18)
android.view.View.performClick(View.java:4508)  
android.view.ViewSPerformClickrunNiew.jave:18575)  
android.os.liandlehhandleCallback(Handlehjave:733)  
android.os.Handler.dispetchMessage(Handlehjava:95)  
android.os.Looperdoop(Looperlaval 30
android.app.ActivityThread.main(ActiviiyThread.java:5590)

nam-van-nguyen, 2020-05-20:
I get this to work by:
1. Get the source from 
https://github.com/square/okio
2. Build it
3. Copy the output .jar file to a new folder called "libs"
4. Update the build.gradle of any failed modules with the dependency:

dependencies {
    implementation fileTree(dir: 'libs', include: ['*.jar'])
...
5. Rebuild

mbhassan, 2020-07-31:
thanks. worked out

closed 2020-07-31 by mbhassan