felHR85/UsbSerial #286: Connecting to multiple serial ports, how can i change the baud rate for each port?

GRSahana, 2019-10-24:
I am using this library in my application, I want to connect to multiple serial port via usb, now in the example code which you have given, How can i change the baud rate for each serial post. Thank You..

ShashiShekhar7, 2019-11-30:
Hey GRSahana!
I'm facing the same issue. Did you got the solution? If yes, then please share. Thanks.

ShashiShekhar7, 2019-11-30:
@GRSahana Hey!
I'm facing the same issue. Did you got the solution? If yes, then please share. Thanks.

GRSahana, 2019-11-30:
You can use the method, serialport.setBaudRate(), to which you will pass
the baudrate as an argument.

ShashiShekhar7, 2019-12-05:
@GRSahana 
I'm already using that method. I think the issue with the service running. Because , once i have selected a baud rate from a selector and started communication, the value of baud rate dose not changes until I re-open the app. Any help? 
PS- I'm using single port to communicate.

ShashiShekhar7, 2019-12-05:
@GRSahana 
I'm already using that method. I think the issue with the service running. Because , once i have selected a baud rate from a selector and started communication, the value of baud rate argument dose not changes until I re-open the app. Any help?
PS- I'm using single port to communicate.

ShashiShekhar7, 2019-12-05:
@GRSahana 
Hey, i just figured out my problem. I just needed to stop the service before changing the baud rate and start again after changing. 
Anyway, thanks for the help.

ShashiShekhar7, 2019-12-05:
@GRSahana
Hey, i just figured out solution to my problem. I just needed to stop the service before changing the baud rate and start again after changing.
Anyway, thanks for the help.

GRSahana, 2019-12-05:
ok👍