felHR85/UsbSerial #290: App is getting hanged when kill the app by user.

imbichitra, 2019-11-05:
HI all developer of UsbSerial i am integrating examplemultipleports for communicating two port the connection and communication with device work fine, but when the user kill the app in middle and open it second time the app is going to hang state just display a white screen , and when i disconnect the Usb and reconnect the device it work fine.

Please help me to solve this problem.

brktrk, 2020-01-07:
Yes, I do.

Shreyas-R, 2020-01-16:
I too encountered this issue while impending multi port example for my requirements. I root caused this issue till usb read function thread. If we use any of inputStream or syncRead in read thread then the app hangs. Just try to comment this read function call, the hang issue goes away. I was unable to get any log during the hang even when I had got strict thread rules enabled for 100% adb logging.

LjieQ, 2020-09-28:
@Shreyas-R How to solve this problem specifically?Can You show me how your code works?, thank you

ahmed-osama-saad, 2024-11-28:
I'm having a similar issue when app is killed while my device is running, the next time i try port!.open() it hangs, I have a timeout around this but the app still hangs.