felHR85/UsbSerial #298: how to distinguish two same machine connected with android(i tried using serial number but getSerialNumber is returning null))

onkarcountrydelight, 2020-01-07:


Shreyas-R, 2020-01-16:
You can use 
```
usbDevice.getDeviceID     // Java
usbDevice.deviceID   //Kotlin 
```
Device ID will always be unique everytime you connect any usb device.

onkarcountrydelight, 2020-01-17:
Thanks for your response, but i want an id which is unique for lifetime of machine (deviceId is getting changed after reconnect the device or restart the app)

Shreyas-R, 2020-01-17:
If you are looking for something like that then you have to either,
(a) manually change the serial number of whatever machine you are interfacing with Android which is currently returning null for you (not defined by machine manufacturer). 
(b) implement some query function within your machine which you have to query when you connect to Android which returns unique ID defined for that machine by you. 

Both require you to tinker with your machine. I don't think there is any other way for that.

onkarcountrydelight, 2020-01-22:
Thanks, will check that