felHR85/UsbSerial #309: ERROR: Failed to resolve: com.github.felHR85:UsbSerial:6.0.4

Harshith996, 2020-04-28:
I have added the following line to the gradle file:
implementation 'com.github.felHR85:UsbSerial:6.0.4'

I have also included maven:
 repositories {
        google()
        jcenter()
        maven { url "https://jitpack.io" }
        
    }

However, I am still getting the 'Failed to Resolve' error with whatever version I try. Please help me fix this issue.

sivaprashanth, 2020-04-28:
Try the latest one which is 6.1.0

Harshith996, 2020-04-28:
> Try the latest one which is 6.1.0

I tried that as well but same error. Is it because I din't include kotlin support?

sivaprashanth, 2020-04-28:
if you are writing a kotlin language program then its necessary otherwise it is not. 

this is my build gradle file -
apply plugin: 'com.android.application'

android {
    compileSdkVersion 26
    useLibrary 'org.apache.http.legacy'
    defaultConfig {
        applicationId "questron.com.v10"
        minSdkVersion 21
        targetSdkVersion 26
        versionCode 1
        versionName "1.0"
        testInstrumentationRunner "android.support.test.runner.AndroidJUnitRunner"
    }

  /*  compileOptions {
        encoding "UTF-8"
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }*/

    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }
    compileOptions {
        sourceCompatibility = 1.8
        targetCompatibility = 1.8
    }
    buildToolsVersion = '25.0.2'
}

dependencies {
    implementation fileTree(dir: 'libs', include: ['achartengine-1.1.0.jar'])
    implementation 'com.android.support:appcompat-v7:26.1.0'
    implementation 'com.android.support.constraint:constraint-layout:1.0.2'
    testImplementation 'junit:junit:4.13'
    androidTestImplementation 'com.android.support.test:runner:1.0.2'
    androidTestImplementation 'com.android.support.test.espresso:espresso-core:3.0.2'
    //implementation 'com.github.mik3y:usb-serial-for-android:Tag'
    implementation 'com.github.felHR85:UsbSerial:6.1.0'

}

you can try this and check or try the examples to get an idea.

Harshith996, 2020-04-28:
Its still not working

mobilinkd, 2020-06-11:
I am having the same problem.
`Failed to resolve: com.github.felHR85:UsbSerial:6.1.0`

Changing the capitalization of the project and repo name to all lower case as shown on the jitpack.io page has no effect.

`buildscript {
	repositories {
		google()
		jcenter()
		mavenCentral()
		maven { url 'https://jitpack.io' }
	}
...
dependencies {
	implementation 'org.scala-lang:scala-library:2.11.12'
	implementation 'com.github.felHR85:UsbSerial:6.1.0'
	implementation fileTree(include: ['*.jar'], dir: 'libs')
}
`

It seems like there is some underlying dependency that is not being met.  Is there some way to diagnose when Android Studio or Gradle is doing under the hood here to see what the actual error message returned when trying to resolve this package is?

mobilinkd, 2020-06-11:
OK, so I found the issue that was affecting me.  There are two places where I needed to specify the jitpack.io dependency.  The second was here:

```
allprojects {
	repositories {
		google()
		jcenter()
		maven { url 'https://jitpack.io' }
	}
}
```

mod157, 2020-07-07:
The same problem occurred and solved.
Add'maven' as the answer above says Build OK
In my case, it was only applied to the Module:Library part in'build.gradle', 
but I added'maven' to gradle: Proejct> repositories.

[image]

Malik056, 2022-01-13:
Just add this in `settings.gradle` file

> maven { url 'https://jitpack.io' }

```
repositories {
        google()
        mavenCentral()
        maven { url 'https://jitpack.io' } // Warning: this repository is going to shut down soon
    }
```

djhunter67, 2024-07-17:
# This COMPILES error FREE
- I successfully compiled with the following settings.
- I am using Kotlin in Android Studio using SDK 34

## build.gradle.kts (:app)
 
dependencies {

    implementation(libs.androidx.core.ktx)
    implementation(libs.androidx.appcompat)
    implementation(libs.material)
    implementation(libs.androidx.activity)
    implementation(libs.androidx.constraintlayout)
    implementation(libs.usbserial)
    testImplementation(libs.junit)
    androidTestImplementation(libs.androidx.junit)
    androidTestImplementation(libs.androidx.espresso.core)
}

## settings.gradle.kts (projectName)
    dependencyResolutionManagement {
        repositoriesMode.set(RepositoriesMode.FAIL_ON_PROJECT_REPOS)
        repositories {
            google()
            mavenCentral()
            maven { setUrl("https://jitpack.io") }
        }
    }

## Conclusion

I have yet to see if the UsbSerial will work but it at least compiles.

fposindia, 2024-10-30:
i solved the issue by updating gradle of my usb_serial which was located at
C:\Users\*****\AppData\Local\Pub\Cache\hosted\pub.dev\usb_serial-0.5.0\android\build.gradle
changed 6.1.0 which is currently have some issue  to 6.0.1

dependencies {
    implementation 'com.github.felHR85:UsbSerial:6.0.6'
}
[image]

harlansr, 2024-11-11:
> i solved the issue by updating gradle of my usb_serial which was located at C:\Users*****\AppData\Local\Pub\Cache\hosted\pub.dev\usb_serial-0.5.0\android\build.gradle changed 6.1.0 which is [...]

Its working for me, thanks to your suggestion

Prince2347X, 2024-11-16:
> i solved the issue by updating gradle of my usb_serial which was located at C:\Users*****\AppData\Local\Pub\Cache\hosted\pub.dev\usb_serial-0.5.0\android\build.gradle changed 6.1.0 which is currently have some issue to 6.0.1
> 
> dependencies { implementation 'com.github.felHR85:UsbSerial:6.0.6' } 

Thanks, worked for me as well

Sitin, 2024-12-22:
Now we have similar problem with UsbSerial `6.1.0`. If you check [https://jitpack.io/com/github/felHR85/UsbSerial/6.1.0/build.log](https://jitpack.io/com/github/felHR85/UsbSerial/6.1.0/build.log) you'll find that build of this library is failed.