felHR85/UsbSerial #313: Looking for Collaborators

felHR85, 2020-06-12:
It's been more than a year since I made a release of UsbSerial and I don't really think that's going to change in the short term.
For that reason I would really welcome  a collaborator(s) to this project. Ideally they should have:
- Good Android knowledge
- Understanding of USB protocols
- Accessibility to the hardware that this project requires (FTDI, CP210x...)

If you are interested please write to **felhr85@gmail.com**

Thank you 😊

sivaprashanth, 2020-06-12:
Hi Felipe

I am interested collaborating to build the USBSerial and i have build many projects with your USB Serial library and have used different chips like arduino, PSOC,Microchip to communicate with the library and i am a software developer and have enormous passion in developing and building the USB Serial library for greater enhancements. 
My linkedin profile : https://www.linkedin.com/in/siva-prashanth-balasubramanian-27249385/

Hope to work with you!

Thanks

Siva Prashanth B
Software Developer 
Questron Technologies Corp
Ph - +1 226 280 4884

felHR85, 2020-06-17:
@sivaprashanth Email sent

tcurdt, 2025-08-07:
What's the status here? Has anyone stepped up?

I would like to use the project but are a little concerned about its future.
@SergioFah seems to have most recent fork but using a low-star fork is also a concern. And it seems to focus on PL2303.