felHR85/UsbSerial #316: FTDI FT232R receiving garbage on read buffer

D4no0, 2020-06-30:
When connecting FT232R (tried cloned analogs and original ones), it receives garbage on read buffer:
```
2020-06-30 13:32:42.077 30128-30177/? I/UsbSerialDebugger: Raw data from Read buffer: 0x01 0x60 
2020-06-30 13:32:42.077 30128-30177/? I/UsbSerialDebugger: Number of bytes obtained from Read buffer: 2
2020-06-30 13:32:42.094 30128-30177/? I/UsbSerialDebugger: Data obtained from Read buffer: `
```

This loops continuously, the async callback never gets called.

Write buffer seems to work fine.

felHR85, 2020-06-30:
What version are you currently using?

D4no0, 2020-06-30:
The latest version, 6.1.0

D4no0, 2020-06-30:
I tried also to use the sync version. It just times out without receiving anything no matter how long the timeout is.

JoshinJoy, 2022-06-28:
I have the same problem. continuously getting 0x01 0x60
this is what is showing in serial terminal of windows.
`````````````````````````````````
it is showing in every baudrate

tr1p1ea, 2023-10-31:
I believe this is a feature of FTDI devices in that it sends out a 'modem status' every 40ms in the absence of data: https://hackage.haskell.org/package/ftdi-0.3.0.2/docs/System-FTDI.html#g:10