felHR85/UsbSerial #322: Add  VID_PID for PL2303GC

HassenLin, 2020-11-03:
PL2303GC's VID_PID is 067B_23A3.

kai-morich, 2020-11-11:
PL2303GC name looks similar to other PL2303 devices, but requires a new driver, so it's unlikely to be supported here by just adding VID/PID. For details see linux kernel driver contributed by prolific: https://patchwork.kernel.org/project/linux-usb/patch/20190924121400.1497-1-charlesyeh522@gmail.com/

HassenLin, 2020-12-11:
@kai-morich  Thank you for your information. After check and porting from kernel driver, it's worked. 

add PL2303GSerialDevice.java and PL2303GIds.java, also need to invoke in UsbSerialDevice.java

=====================
PL2303GIds.java
=====================
package com.felhr.deviceids;

import static com.felhr.deviceids.Helpers.createTable;
import static com.felhr.deviceids.Helpers.createDevice;

public class PL2303GIds
{
    private PL2303GIds()
    {

    }

    private static final long[] pl2303GDevices = createTable(
            createDevice(0x067b, 0x23a3)
    );

    public static boolean isDeviceSupported(int vendorId, int productId)
    {
        return Helpers.exists(pl2303GDevices, vendorId, productId);
    }
}

=====================
PL2303GSerialDevice.java
=====================

package com.felhr.usbserial;

import android.hardware.usb.UsbConstants;
import android.hardware.usb.UsbDevice;
import android.hardware.usb.UsbDeviceConnection;
import android.hardware.usb.UsbEndpoint;
import android.hardware.usb.UsbInterface;
import android.hardware.usb.UsbRequest;
import android.util.Log;

import com.felhr.utils.SafeUsbRequest;

public class PL2303GSerialDevice extends UsbSerialDevice
{
    private static final String CLASS_ID = PL2303GSerialDevice.class.getSimpleName();

    private static final int PL2303_REQTYPE_HOST2DEVICE_VENDOR = 0x40;
    private static final int PL2303_REQTYPE_DEVICE2HOST_VENDOR = 0xC0;
    private static final int PL2303_REQTYPE_HOST2DEVICE = 0x21;

    private static final int PL2303_VENDOR_READ_REQUEST = 0x01;
    private static final int PL2303_VENDOR_READ_NREQUEST = 0x81;
    private static final int PL2303_SET_LINE_CODING = 0x20;
    private static final int PL2303_SET_CONTROL_REQUEST = 0x22;
    private static final int PL2303_READ_TYPE_HX_STATUS = 0x8080;
    private static final int PL2303_HXN_RESET_REG = 0x7;
    private static final int PL2303_HXN_RESET_UPSTREAM_PIPE	= 0x2;
    private static final int PL2303_HXN_RESET_DOWNSTREAM_PIPE = 0x1;
    private final byte[] defaultSetLine = new byte[]{
            (byte) 0x80, // [0:3] Baud rate (reverse hex encoding 9600:00 00 25 80 -> 80 25 00 00)
            (byte) 0x25,
            (byte) 0x00,
            (byte) 0x00,
            (byte) 0x00, // [4] Stop Bits (0=1, 1=1.5, 2=2)
            (byte) 0x00, // [5] Parity (0=NONE 1=ODD 2=EVEN 3=MARK 4=SPACE)
            (byte) 0x08  // [6] Data Bits (5=5, 6=6, 7=7, 8=8)
    };
[... 335 lines cut ...]
        // End of specific vendor stuff
        if(setControlCommand(PL2303_REQTYPE_HOST2DEVICE, PL2303_SET_CONTROL_REQUEST, 0x0003, 0,null) < 0)
            return false;
        if(setControlCommand(PL2303_REQTYPE_HOST2DEVICE, PL2303_SET_LINE_CODING, 0x0000, 0, defaultSetLine) < 0)
            return false;
        if(setControlCommand(PL2303_REQTYPE_HOST2DEVICE_VENDOR, PL2303_VENDOR_READ_NREQUEST, 0x0505, 0x1311, null) < 0)
            return false;
*/
        setControlCommand(PL2303_REQTYPE_DEVICE2HOST_VENDOR, PL2303_VENDOR_READ_REQUEST, PL2303_READ_TYPE_HX_STATUS, 0, buf); // TODO. check TYPE_1 or TYPE_HN

        setControlCommand(PL2303_REQTYPE_HOST2DEVICE_VENDOR, PL2303_VENDOR_READ_NREQUEST, PL2303_HXN_RESET_REG, PL2303_HXN_RESET_UPSTREAM_PIPE | PL2303_HXN_RESET_DOWNSTREAM_PIPE, null);

        return true;
    }

    private int setControlCommand(int reqType , int request, int value, int index, byte[] data)
    {
        int dataLength = 0;
        if(data != null)
            dataLength = data.length;
        int response = connection.controlTransfer(reqType, request, value, index, data, dataLength, USB_TIMEOUT);
        Log.i(CLASS_ID,"Control Transfer Response: " + String.valueOf(response));
        return response;
    }
}

closed 2020-12-11 by HassenLin