felHR85/UsbSerial #339: Failed to resolve: com.github.felHR85:UsbSerial:6.1.0 when move to Artic Fox

jorgeEA3HKZ, 2021-09-07:
Hi,
I upgrade to Artic Fox 2020.3.1 Patch2 (built Aug 26, 2021)
the error resolving UsbSerial appears:
Failed to resolve: com.github.felHR85:UsbSerial:6.1.0
Project Gradle:
buildscript {
    repositories {
        google()
        mavenCentral()
        maven { url "https://jitpack.io" }
    }
    dependencies {
        classpath "com.android.tools.build:gradle:7.0.2"
        classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:1.5.30"
        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}
task clean(type: Delete) {
    delete rootProject.buildDir
}

and Module Gradle:

plugins {
    id 'com.android.application'
    id 'kotlin-android'
}
android {
    compileSdk 30
    defaultConfig {
        applicationId "com.jcastano.testserialport_1"
        minSdk 29
        targetSdk 30
        versionCode 1
        versionName "1.0"
        testInstrumentationRunner "androidx.test.runner.AndroidJUnitRunner"
    }
    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android-optimize.txt'), 'proguard-rules.pro'
        }
    }
    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }
    kotlinOptions {
        jvmTarget = '1.8'
    }
}
dependencies {
    implementation 'androidx.core:core-ktx:1.6.0'
    implementation 'androidx.appcompat:appcompat:1.3.1'
    implementation 'com.google.android.material:material:1.4.0'
    implementation 'androidx.constraintlayout:constraintlayout:2.1.0'
    testImplementation 'junit:junit:4.+'
    androidTestImplementation 'androidx.test.ext:junit:1.1.3'
    androidTestImplementation 'androidx.test.espresso:espresso-core:3.4.0'
    implementation 'com.github.felHR85:UsbSerial:6.1.0'
}

Regards

UnisonControls, 2021-11-17:
Facing same issue in Android Studio Arctic Fox | 2020.3.1 Patch 3
@felHR85  can you please help us here.

farmisen, 2022-06-11:
The config bellow works for me. 

**in root build.gradle**:
```
buildscript {
    ...
    repositories {
        google()
        mavenCentral()
        maven { url "https://jitpack.io" }
    }
    ...
}
```

**in settings.gradle**:
```
dependencyResolutionManagement {
    repositoriesMode.set(RepositoriesMode.FAIL_ON_PROJECT_REPOS)
    repositories {
        google()
        mavenCentral()
        maven { url "https://jitpack.io" }
    }
}
```

**in module build.gradle**:
```
dependencies {
    implementation 'com.github.felHR85:UsbSerial:6.1.0'
}
```