felHR85/UsbSerial #349: getDeviceList empty

aprldntm1234, 2022-02-16:
I developed a program using this library.
        HashMap<String, UsbDevice> usbDevices = usbManager.getDeviceList();
        Log.d("usbDevices_log", usbDevices.toString());

where usbManager.getDeviceList(); When I output the value as log, it came out empty.

Is there no solution? For reference, the current version is 5.1.1 Lollipop.

[image]

davdavcoco, 2022-11-04:
hey, did your problem solved?

eclectice, 2022-11-04:
Did you use an Android emulator that Android Studio provided to run your app? In that case, It is not supported.

https://developer.android.com/studio/run/advanced-emulator-usage#limitations

Other than that, try to check USB Host support for your hardware device with this app from Google Play Store: https://play.google.com/store/apps/details?id=eu.chainfire.usbhostdiagnostics