felHR85/UsbSerial #360: Issue with SDK > 31 on Android 12

bdureau, 2023-02-04:
Looks like it does not request the permission correctly on Android 12.
You might be able to fix it like this
This is on line 155 on file SerialPortBuilder.java 
//PendingIntent mPendingIntent = PendingIntent.getBroadcast(context, 0, new Intent(ACTION_USB_PERMISSION), 0);
        PendingIntent mPendingIntent = null;
        if(android.os.Build.VERSION.SDK_INT >= 31) {
            mPendingIntent = PendingIntent.getBroadcast(context, 0, new Intent(ACTION_USB_PERMISSION), PendingIntent.FLAG_IMMUTABLE);
        } else {
            mPendingIntent = PendingIntent.getBroadcast(context, 0, new Intent(ACTION_USB_PERMISSION), 0);
        }

Unfortunately I am having trouble to build the library so I cannot really test it.
I am quite happy to confirm if it works if someone explain how to build the lib
Regards
Boris

bdureau, 2023-02-19:
I have managed to recompile the library and it works
however I have had to add the following dependency for lib serial to work
implementation 'com.squareup.okio:okio:3.3.0'

worldhugo, 2025-01-10:
Hi !

Can you share your lib version ?

bdureau, 2025-01-10:
mine is a bit messy but you could use the following fork which I am now using https://github.com/BeebLaser/UsbSerial