felHR85/UsbSerial #369: Host and Accessory communication(NOT AN ISSUE)

HafizMSaad, 2024-02-01:
I'm encountering difficulties establishing communication between two Android devices, where one is designated as the host and the other as the accessory. I attempted to utilize the usb-serial-for-android library on one device and connected the two devices using a wired connection. However, I'm facing issues in establishing a connection and facilitating communication between the devices.

janlu-pretorius, 2024-05-16:
Did you find a solution? I have a similar problem I am trying to solve. In my case I can detect device A from device B, but I can't detect device B from device A.

HafizMSaad, 2024-05-16:
> Did you find a solution? I have a similar problem I am trying to solve. In my case I can detect device A from device B, but I can't detect device B from device A.

Nope, I haven't. 
So, I left it and went for other solutions like TCP/UDP.