felHR85/UsbSerial #375: The latest version can't be accessed from Jitpack anymore

ChaseGuru, 2024-10-22:
For whatever reason, it seems like the cached build of 6.1.0 expired and jitpack ended up attempting to rebuild it on the 18th.

But that failed:
https://jitpack.io/com/github/felHR85/UsbSerial/6.1.0/build.log

I referenced the latest commit hash instead to force jitpack to build that (since it doesn't seem to be retrying for 6.1.0)

That build artifacts successfully:
https://jitpack.io/com/github/felHR85/UsbSerial/7ad6c9f6/build.log

Except, the artifacts seem to be different than before, and projects asking for the dependency find the package on jitpack, but still can't find the artifacts, just for different reasons now.  

App's using:
```
implementation 'com.github.felHR85:UsbSerial:7ad6c9f6'
```
Will fail to find com/github/felHR85/UsbSerial/7ad6c9f6/UsbSerial-7ad6c9f6.aar - correctly, because the actual artifact was:
com/github/felHR85/UsbSerial/7ad6c9f6/UsbSerial-7ad6c9f6-release.aar

I managed to specify a classifier which seemed to move the error to the .jar.

markbreuss, 2024-10-23:
Any update on this or does anyone know a suitable workaround?

This is currently blocking our Release Pipeline.

lugia2491, 2024-10-24:
You can import .aar file to temporary solution fix the problem.
Download and put 6.1.0 aar file to project/lib
implementation the aar:
`implementation files('../libs/usbserial-6.1.0-release.aar')`

Aleksandar98, 2024-10-24:
Switching back from 6.1.0 to version 6.0.6 worked for me.

hwh97, 2024-10-28:
> Will fail to find com/github/felHR85/UsbSerial/7ad6c9f6/UsbSerial-7ad6c9f6.aar - correctly, because the actual artifact was: com/github/felHR85/UsbSerial/7ad6c9f6/UsbSerial-7ad6c9f6-release.aar

@ChaseGuru Hey, you can try below custom implementation, it works on my side.
```
implementation('com.github.felHR85:UsbSerial:7ad6c9f6') {
    artifact {
        name = 'UsbSerial'
        classifier = 'release'
        type = 'aar'
    }
}
```

angelix, 2024-11-04:
> @ChaseGuru Hey, you can try below custom implementation, it works on my side.
>
> ```
> implementation('com.github.felHR85:UsbSerial:7ad6c9f6') {
>     artifact {
>         name = 'UsbSerial' [...]

On the same principle i created a MR to update the dependencies and build successfully on jitpack.

MR: https://github.com/felHR85/UsbSerial/pull/377
Dependency:
```
dependencies {
      implementation 'com.github.angelix:UsbSerial:fdefe94f65'
}
```

ChaseGuru, 2024-11-07:
> @ChaseGuru Hey, you can try below custom implementation, it works on my side.
>
> ```
> implementation('com.github.felHR85:UsbSerial:7ad6c9f6') {
>     artifact {
>         name = 'UsbSerial' [...]

Thanks, I'll try it now.  I tried a configured implementation like that, but I didn't find the right combination of variables to get it to use the release artifact correctly.

ChaseGuru, 2024-11-07:
> version

This is the workaround I did to get my deployment out when I was messing with it originally.

ChaseGuru, 2024-11-07:
> Switching back from 6.1.0 to version 6.0.6 worked for me.

This is the workaround I did originally to get my deploy out.

jrzerr, 2024-11-18:
> @ChaseGuru Hey, you can try below custom implementation, it works on my side.
>
> ```
> implementation('com.github.felHR85:UsbSerial:7ad6c9f6') {
>     artifact {
>         name = 'UsbSerial' [...]

thanks this worked perfectly

ChaseGuru, 2024-11-18:
> thanks this worked perfectly

Yeah, there's a lot of great work arounds, and I've got it working. I'm using a flutter package, and someone made a different workaround in a fork of that which is what I'm currently using.

My favorite fix probably is https://github.com/felHR85/UsbSerial/pull/377 - since it upgrades stuff and doesn't require changes to how the library is imported (if merged into this repo nobody would need to migrate their dependency declaration).

ChaseGuru, 2024-11-18:
The fork I'm using was this fork:
https://github.com/altera2015/usbserial/pull/119

Which is actually using the fix you suggested, as it was written by [hwh97](https://github.com/hwh97) who also made the initial suggestion.  Though as mentioned, I'd probably prefer using #377's fork just because it upgrades stuff and seems like the best long term solution.

angelix, 2024-11-19:
> The fork I'm using was this fork: [altera2015/usbserial#119](https://github.com/altera2015/usbserial/pull/119)
> 
> Which is actually using the fix you suggested, as it was written by [hwh97](https://github.com/hwh97) who also made the initial suggestion. Though as mentioned, I'd probably prefer using #377's fork just because it upgrades stuff and seems like the best long term solution.

My latest commit '0f16336a4e' fixed the build issue related to aar naming. Seems that 'archivesBaseName' screwed things.

PR https://github.com/felHR85/UsbSerial/pull/377
```
dependencies {
      implementation 'com.github.angelix:UsbSerial:0f16336a4e'
}
```

jose-silva-wisecrop, 2024-11-21:
On my flutter project I stil get this * What went wrong:
Execution failed for task ':app:checkDebugAarMetadata'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Failed to transform UsbSerial-7ad6c9f6.aar (com.github.felHR85:UsbSerial:7ad6c9f6) to match attributes {artifactType=android-aar-metadata, org.gradle.category=library, org.gradle.dependency.bundling=external, org.gradle.libraryelements=aar, org.gradle.status=release, org.gradle.usage=java-runtime}.
      > Could not find UsbSerial-7ad6c9f6.aar (com.github.felHR85:UsbSerial:7ad6c9f6).
        Searched in the following locations:
            https://jitpack.io/com/github/felHR85/UsbSerial/7ad6c9f6/UsbSerial-7ad6c9f6.aar

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 32s
Error: Gradle task assembleDebug failed with exit code 1

josephmangmang, 2024-11-25:
Missing debug aar
https://jitpack.io/com/github/felHR85/UsbSerial/9b915b2f84/

ErMapsh, 2024-11-27:
> You can import .aar file to temporary solution fix the problem. Download and put 6.1.0 aar file to project/lib implementation the aar: `implementation files('../libs/usbserial-6.1.0-release.aar')`

As per your solution, If I put the .aar in local directly then while building the application got error  **Direct local .aar file dependencies are not supported when building an AAR**

josephmangmang, 2024-11-27:
Fixed with 
dependencies {
      implementation 'com.github.angelix:UsbSerial:0f16336a4e'
}